micro-ROS/micro-ROS.github.io #20: Porting to freeRTOS

tiagoamaralp, 2019-03-21:
Do u have any tips or documentation so that I can port to freeRTOS?

ralph-lange, 2019-04-15:
Great that you are interested in porting micro-ROS to a different RTOS. Currently, we are focused on NuttX and more generally on POSIX-based RTOS. Therefore, I propose that you start with a port to FreeRTOS+POSIX (cf. https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_POSIX/), which is still under development but already available at https://github.com/aws/amazon-freertos/tree/master/lib/FreeRTOS-Plus-POSIX. Unfortunately, I have no hands-on experience with Free-RTOS myself.

By the way, Micro-XRCE-DDS aims at supporting a larger spectrum of RTOS in the long-term. @BorjaOuterelo, can you provide specific information on this?

Please keep up updated. Such a port would be awesome!

BorjaOuterelo, 2019-04-15:
Hi @tiagoamaralp 

As @ralph-lange mentioned the plans for [Micro XRCE-DDS](https://github.com/eProsima/Micro-XRCE-DDS) is to support more RTOS in the long-term. We have designed it to be easily portable to different systems and as it is Micro XRCE-DDS right now the port to different platforms should not be tricky if you create custom transports.
Apart from custom transports, you may need to change the time implementation: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/util/time.c

Regarding documentation on these custom transports you can find it in a branch in Micro XRCE-DDS: https://github.com/eProsima/Micro-XRCE-DDS-docs/blob/feature/transport/docs/transport.rst. This branch will be incorporated into the main documentation branch for the next release.

Regarding micro-ROS either you try the FreeRTOS+POSIX  or create a custom transport in the middleware (Micro XRCE-DDS) and try to run micro-ROS on top of it. Haven't tried the latter personally but ideally, the dependencies to the RTOS should be at that level primarily.

closed 2022-01-28 by pablogs9