micro-ROS/micro-ROS.github.io #52: Overview Section Rework

imuguruza, 2019-07-10:
### [Overview → Overview](https://micro-ros.github.io/docs/overview/) [EPROS]

Rename to “Overview → Architecture” or “Overview → Architectural Overview”
Remove the paragraph on tutorials as the link to tutorials is now placed prominently in the top navigation bar
Update architecture diagram with the colored ones from Deliverable D1.5
Revise lists of concepts and source code repos
Add matrix indicating support of ROS 2 releases in the style of:

Prominently link to .repos file

### [Overview → Hardware](https://micro-ros.github.io/docs/overview/hardware/) [ALR]
Add paragraph explaining that our stack can be brought to a number of platforms. Argue about use of Posix standard and point to large number of platforms supported by NuttX.

### [Overview → Comparison to other approaches to put ROS onto small embedded devices](https://micro-ros.github.io/docs/overview/comparison/) [BOSCH]
Shorten title: What about “Comparison to other embedded ROS approaches”?
List and link further known approaches, e.g., mROS

imuguruza, 2019-07-10:
> Overview → Hardware [ALR]
>Add paragraph explaining that our stack can be brought to a number of platforms. Argue about use of Posix standard and point to large number of platforms supported by NuttX.

I am thinking that makes more sense to add the explanations under "Concepts => RTOS", we could just put a small list and the link to the complete list at this section. Otherwise, we would mix concept explanation and the supported board list

imuguruza, 2019-07-10:
>Overview → Hardware [ALR]
>Add paragraph explaining that our stack can be brought to a number of platforms. Argue about use of Posix standard and point to large number of platforms supported by NuttX.

Opened a [PR](https://github.com/micro-ROS/micro-ROS.github.io/pull/56) to improve it

imuguruza, 2019-07-10:
Closed previous one an opened [a new on](https://github.com/micro-ROS/micro-ROS.github.io/pull/58)e, based on the new master that contains no-broken links

ralph-lange, 2019-08-02:
Made large improvements to start page and main overview page in PR https://github.com/micro-ROS/micro-ROS.github.io/pull/63. Further changes to overview page needed by @julianbermudez et al., cf. this issue.

ralph-lange, 2020-03-05:
Superseded by https://github.com/micro-ROS/micro-ROS.github.io/issues/107

closed 2020-03-05 by ralph-lange