micro-ROS/micro-ROS.github.io #54: Tutorials Section Rework

imuguruza, 2019-07-10:
+ [Basic Tutorials → Getting Started with Your Computer](https://micro-ros.github.io//docs/tutorials/basic/getting_started/) [EPROS/ALR]
  + Remove “(WIP)” from page title
  + Shorten the lengthy headings to one line
  + Revise and complete texts or link to instructions in the corresponding repo
    + I prefer having the instructions in the repo and have a short, generic description on the webpage only.
  + Move examples at the end of the page to separate page (or in case of the executor to the corresponding page in the concepts section)

+ [Basic Tutorials → Getting Started with an Embedded Board](https://micro-ros.github.io/docs/tutorials/basic/getting_started_embedded/) [EPROS/ALR]
  + Remove “(WIP)” from page title
  + ToC not needed for this short page.
  + Remove “Introduction” heading and start with text directly.
  + Complete the text, cf. “To be completed at the end”
  + Add screenshot or photo?

+ [Basic Tutorials → Getting Started with simulator](https://micro-ros.github.io/docs/tutorials/basic/getting_started_simulator/) [ALR]
  + Remove “(WIP)” from page title
  + Complete the page
  + Add screenshot or photo?

+ [Advanced Tutorials → NuttX Getting Started](https://micro-ros.github.io/docs/tutorials/advanced/nuttx_getting_started/) [ALR]
  + Add ToC at beginning of page

+ Advanced Tutorials → Benchmarking [PIAP]
  + Create new page.

+ Create new page for TB2 demo [@ralph-lange ]
  + Should we create a separate “demo submenu”? Yes.

imuguruza, 2019-07-11:
@BorjaOuterelo I have been checking the "Basic Tutorials" and I suggest to make a basic landing page where we list all the getting started tutorials.

We could briefly explain for what is each and in each repository add the required instructions to execute them. I would add in each "client" docker the agent as well, so in the same docker file client and agents are placed.

Otherwise you need to give Docker to host computer resources, and share with the client Docker, which I think is too complex.

imuguruza, 2019-07-11:
What's your opinion?

BorjaOuterelo, 2019-07-11:
@imuguruza. so to follow the tutorials user should head to a repository? Can you provide an example?.

The Docker solution, I would like to investigate the docker compose option it seems a good match to a multiplication solution (client + agent). As a first glance, it seems that we need just to create a "tutorial" compose file. using existing dockers.

imuguruza, 2019-07-11:
> @imuguruza. so to follow the tutorials user should head to a repository? Can you provide an example?.

Yeah @BorjaOuterelo , something like [this](https://github.com/micro-ROS/micro-ROS-bridge_RPI/tree/master/RPI_6lowpan). At each repo we should explain the steps to compile if required and how to execute it

BorjaOuterelo, 2019-07-11:
I am OK with either option as you mention is easier to maintain this link way, however it would look like an empty website.
There are some tutorials in https://github.com/micro-ROS/micro-ROS-doc/tree/develop that I think could be really useful here, like the getting started and demos one. So we should provide a link to them?

iluetkeb, 2019-09-19:
https://github.com/micro-ROS/micro-ros-build/issues/18#issuecomment-532673042 has some hints for improvement as well.

ralph-lange, 2020-03-05:
Superseded by https://github.com/micro-ROS/micro-ROS.github.io/issues/109

closed 2020-03-05 by ralph-lange