micro-ROS/micro-ROS.github.io #109: Rework Tutorials Section

ralph-lange, 2020-03-05:
- [ ] Change "Basic tutorials" to "Core tutorials" with pages on
    - [ ] Hello World with micro-ROS (simple echo node with topics /ping and /pong)
    - [ ] Programming with rcl and rclc @JanStaschulat 
    - [ ] rclc Executor @JanStaschulat 
    - [ ] Benchmarking micro-ROS applications (put "Tracing with ROS 2" here inside? Does this make sense?) @amx-piap 
    - [ ] Setting up the rmw for Micro-XRCE-DDS (incl. best-effort vs reliable) @julibert 

- [ ] Improve demo subsection. Finally should be five pages on
    - [ ] Kobuki Demo (NuttX)
    - [ ] Crazyflie Demo (FreeRTOS)
    - [ ] ToF Sensor (Zephyr)
    - [ ] Interfacing with the FIWARE Context Broker (ToF Sensor via Context Broker)
    - [ ] Joint Kobuki - Crazyflie - ToF Sensor - FIWARE Context Broker Demo

- [ ] Introduce new Advanced Tutorials with NuttX subsection. Organize pages into
    - [ ] NuttX in Depth - split existing page up
    - [ ] Debugging a NuttX target (incl. VS Code)
    - [ ] 6LoWPAN with NuttX
    - [ ] Simulator (if possible)
    - [ ] Adding Micro-ROS to a NuttX board configuration

- [ ] Introduce new empty subsection named "Advanced Tutorials with FreeRTOS"

- [ ] Introduce new subsection named "Advanced Tutorials with Zephyr"
    - [ ] Create tutorial page on Zephyr simulator

Other issues:
- [ ] Where to put "Tracing with ROS 2" if not integrated in benchmarking page?

ralph-lange, 2020-03-05:
More details on the Hello World tutorial:
- Assume basic knowledge of ROS 2 and that ROS 2 is installed on desktop/laptop
- Include code for echo node on the MCU with topics /ping and /pong
- Should be one tutorial page only, but possibly with RTOS-specific paragraphs
- Echo function to be tested by user from ROS 2 on the command line

amx-piap, 2020-03-06:
Regarding the Ros2 Tracing and benchmarking , we can have seperated section:
 -   Tracing:
                  - ROS2 Tracing etc..
                  - Tracing in Benchmarking (more information how to exploit traces for benchmarks).
 -   Benchmarking (Dedicated section about benchmarking)

FranFin, 2020-11-11:
@ralph-lange Closing this issue as the pages referring to the demos have been created, even though content is missing.

closed 2020-11-11 by FranFin