micro-ROS/micro-ROS.github.io #141: Trouble building micro-ROS agent

mik268, 2020-04-13:
Hello,

I have followed the instructions for NuttX getting started. Everything goes well until attempting to build the micro-ROS agent. 

# Download micro-ROS-Agent packages
ros2 run micro_ros_setup create_agent_ws.sh

# Build micro-ROS-Agent packages, this may take a while.
colcon build --metas src

The last command attempts to build but returns "error: 'ucdrBuffer' has no member named 'offset'"
This returns "Failed <<< microxrcedds_agent [Exited with code 2]

I have tried several things including downloading the microROS agent from its github page but it has not worked. Any suggestion?

mik268, 2020-04-23:
Wanted to follow up and see if there were any suggestions for installing MicroROS Agent and where to  build it. Should it be contained within a workspace or can it be called from any workspace? The current instructions build using microros build, within a specified workspace.

pablogs9, 2020-04-24:
Please check again an installation because there has been an active development during last days. It is possible to install an agent in any workspace, but I would recommend having an isolated workspace for the agent by now.

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9