micro-ROS/micro-ROS.github.io #143: Restructuring of tutorials

ralph-lange, 2020-04-16:
Core Tutorials
- [ ] "First micro-ROS application on Linux" @julibert, @FranFin 
    * Assumption: First Linux laptop even without ROS.
    * Point to Dockerfile at https://github.com/micro-ROS/docker/blob/dashing/base/Dockerfile 
    * Steps to install ROS, micro-ROS do not have to be explained, but one sentence + link to instructions is enough
    * Explain how to setup micro-ROS build tooling
    * To create this page use https://micro-ros.github.io/docs/tutorials/advanced/linux/linux_getting_started/ as a basis and move it accordingly.
    * Explain how to compile and run ping pong application.
    * Explain the most important lines of code in this application and then refer to the separate "Programming with rcl and rclc tutorial".
- [ ] "First micro-ROS application on an RTOS" @julibert , @FranFin 
    * Explain commands line for building and flashing by some "hello world"-like example for all three RTOS (by merging the contents from the FreeRTOS Getting Started and Zephyr Getting Started)
    * Explain how to connect the board JTAG etc. (cf. https://micro-ros.github.io/docs/tutorials/advanced/nuttx/udp_echo/)
    * Use ping-pong app as the first application
    * Explain that our build system covers three RTOS
    * Finally refer to https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup for more information.
    * Create this tutorial using https://micro-ros.github.io/docs/tutorials/advanced/nuttx/nuttx_getting_started/ as a starting point but move it to the core tutorials section.
    * Also merge contents from https://micro-ros.github.io/docs/tutorials/advanced/nuttx/add_microros_config/ and then remove this page.
- [ ] "Programming with rcl and rclc"  @ralph-lange, @JanStaschulat, @norro  
    * Explain that we take most ROS 2 concepts to micro-ROS
    * Explain how we build upon existing ROS 2 layers (rmw, rcl)
    * Explain rclc package with convenience functions
    * Explain that rclcpp might be also used on stronger MCUs (but not in scope of this tutorial)
    * Table of contents
    * Section on topics (pub/sub)
    * Section on services
    * Section on timers
    * Section on the rclc Executor
    * Section on parameters
    * Section on lifecycle (aka managed nodes)
- [ ] "Optimize middleware configuration"  @julibert 
    * Make this page a real tutorial.
    * Use ping-pong application as the running example (e.g., set number of services to zero)
    * Remove list of all configuration parameters but refer either to some README.md in the corresponding repo or move the list to the concepts section of the website.
    * Explain step by step where and how to change the configuration (in cmake).

Advanced Tutorials
- [ ] "NuttX NSH console over USB or UART"  @jfm92 
    * Merge the two existing tutorials
    * Shorten description how to build the application but refer to core tutorials  for these steps.
- [ ] "6LoWPAN with micro-ROS on NuttX"  @jfm92 
    * Merge the existing two pages on 6LoWPAN
- [ ] "Debugging a NuttX target"  @tkolcon 
    * Merge the two existing tutorials
- [ ] "Adding Micro-ROS to a NuttX board configuration"  @ralph-lange 
    * Just move to advanced tutorials section
- [ ] "Zephyr Emulator"   @pablogs9 
    * State that again the ping-pong app is being used as a running example and that we assume that the reader knows how to build this app for Zephyr from the "First micro-ROS application on an RTOS".
    * Shorten this tutorial and only explain the steps that are different compared with the "First micro-ROS application on an RTOS"
- [ ] "Getting started with ROS 2 tracing"   @ralph-lange 
    * Just move to advanced tutorials section
- [ ] "Some tutorial on benchmarking"  @amx-piap 

Further TODOs
- [ ] Remove udp and tcp server tutorials   @ralph-lange

FranFin, 2020-11-11:
@ralph-lange: Closing this issue as the only task remaining is to create a "Section on parameters (who?)", and this feature is not implemented yet.

closed 2020-11-11 by FranFin

ralph-lange, 2020-11-11:
@FranFin, yes, as soon as parameters have been implemented, a corresponding section should be added to https://micro-ros.github.io/docs/tutorials/core/programming_rcl_rclc/.
In my opinion there is no need to document this in an issue here as it should be normal to always add a section to this tutorial when implementing new features in rclc.