micro-ROS/micro-ROS.github.io #157: First micro-ROS Application on an RTOS: make menuconfig fails

Kelly-Coffey, 2020-05-03:
From a ROS2 installation all micro-ROS Nuttx tutorials fail the make menuconfig step  since there are no tutorial steps how to install kconfig-frontends tools.
I have verified this with my own ROS2 installation and the docker "ros:dashing". 

The make menuconfig works in the "micro-ROS-Olimex-NuttX:dashing" docker .

It would be helpful to include instructions how to install kconfig-frontends for the Nuttx tutorials where menuconfig will be used.

There is also a typo in the Application development section
"Now we’re going to considerer the folder application the next folder: uros_ws/firmware/apps/examples/uros_ws/firmware/apps/examples"

Should it be ?
"Now we’re going to consider the application folder: uros_ws/firmware/apps/examples/uros_pingpong"

pablogs9, 2020-05-04:
Addressed here: https://github.com/micro-ROS/micro-ROS.github.io/pull/149

Please close when merged.

Kelly-Coffey, 2020-05-21:
Closing now 149 is merged. I verified the kconfig install instructions are present.

closed 2020-05-21 by Kelly-Coffey