micro-ROS/micro-ROS.github.io #167: Refactor 'First micro-ROS Application on an RTOS' tutorial

ralph-lange, 2020-05-26:
- Create separate pages (urls) for each RTOS (NuttX, FreeRTOS, and Zephyr)
- Ensure redirect from '/docs/tutorials/advanced/zephyr/zephyr_getting_started/'
- Avoid three completely separate pages but use includes to generate the pages from common sections
- Introduce a nice overview page to ask the reader to decide for one RTOS for this tutorial.

ralph-lange, 2020-05-26:
@FranFin, no need to use Jinja2 for generating the three versions of the 'First micro-ROS application on an RTOS' from a common template. Jekyll provides include directives for this. I first tried [include_relative](https://jekyllrb.com/docs/includes/#including-files-relative-to-another-file), which works with Markdown and HTML, but there seems to be a [bug](https://talk.jekyllrb.com/t/include-relative-could-not-locate-the-included-file/3148) in the version 3.8.5.
As a fallback solution, I switched to a normal include, which refers to the `_includes` folder.

ralph-lange, 2020-05-26:
I've pushed my very first changes to https://github.com/micro-ROS/micro-ROS.github.io/tree/feature/rtos-specific_tutorials_with_common_sections.

ralph-lange, 2020-05-26:
Relates to https://github.com/micro-ROS/micro-ROS.github.io/issues/143.

dayjaby, 2020-06-14:
+ Sample app link is broken for NuttX

ralph-lange, 2020-06-30:
Fixed in https://github.com/micro-ROS/micro-ROS.github.io/commit/57dfe9f11ccaf90fff576d6ee317507bd9a2a666

closed 2020-11-11 by ralph-lange