micro-ROS/micro-ROS.github.io #177: Procedure for porting micro-ROS to new boards?

peterpolidoro, 2020-07-07:
I was not sure where to ask this question, I apologize if this is the wrong place.

Do you have any more detailed instructions for porting micro-ROS to new boards? In the documentation you have some links to porting NuttX. Do you have more general instructions for porting to other another RTOS as well? Do you have a list of all the repositories that would need modification? I am happy to help with time and effort to this port, but I need a little guidance first on how to best contribute.

I would like to run a micro-ROS Dashing client on a [Teensy 4](https://www.pjrc.com/store/teensy40.html). It uses a NXP iMXRT1062 processor which you list as a support MCU. Does that just mean it is supported by NuttX?

Someone has ported FreeRTOS to run on the Teensy 4 [FreeRTOS-Teensy4](https://github.com/discord-intech/FreeRTOS-Teensy4/tree/master).

Is it possible to port it in such a way that Arduino libraries and C++ code will work with rclcpp as a micro-ROS client on the Teensy board? Is this relatively quick and easy or would that take significant effort to make it work? Thanks!

pablogs9, 2020-07-08:
Hello @peterpolidoro, by now there is no standard procedure for porting micro-ROS to other boards but you can check the efforts of some other people for example:

- https://github.com/micro-ROS/micro-ros-build/pull/139
- https://github.com/micro-ROS/freertos_apps/pull/11
- https://github.com/micro-ROS/freertos_apps/issues/7

In general you should be able to integrate your build system in micro-ROS build system four-step procedure. Then, in the `freertos_apps` repo you can integrate all the required companion code.

In order to port the full micro-ROS stack you would need some POSIX compliance, a C++11 compiler and XRCE-DDS transports. The former should be straight forward if you have POSIX compliance for serial devices or UDP/IP stack.

Feel free to ask if you have any doubts during the process and PR your achievements!

Thanks!

ppisa, 2020-12-06:
NuttX is ported to Teensy 4.0 and Teensy 4.1

https://github.com/apache/incubator-nuttx/tree/master/boards/arm/imxrt/teensy-4.x

The BSP is provided by Michal Lenc https://github.com/michallenc/

We have even baseboard for Teensy 4.1 with 3x CAN FD capable transceivers, RS232, RS485, small LCD display for setup, stepper, BDLC and DC motor control in plastic DIN rail compatible enclosure.

The design of base board is open
https://gitlab.com/pikron/projects/imxrt-devel/-/tree/master/hw

The design is created in PEDA schematic+PCB software
https://sourceforge.net/projects/peda/

closed 2021-01-06 by peterpolidoro

ppisa, 2021-01-06:
For your information, Teensy 4.0 and 4.1 support is included in NuttX mainline now

Pull request https://github.com/apache/incubator-nuttx/pull/2463

BSP directory in the official NuttX repository

https://github.com/apache/incubator-nuttx/tree/master/boards/arm/imxrt/teensy-4.x

serial, SPI, CAN, CAN-FD, Ethernet, console over telnet, etc... is working. Even CDC-ACM console seems to work but there are some issues still when I have tested it last time. Michal Lenc works on better coexistence of CAN only and CAN-FD capable interfaces, imxRT has two interfaces CAN only and one with CAN-FD option and we want to use same driver with common configuration for all. 

If you have interest to use BSP and even cooperate on imxRT support, contact me.

If you consider to port some ROS project to RTEMS https://www.rtems.org/ for some serious applications (European Space Agency, NASA) then contact me as well. I am in contact with people from ESA and I know RTEMS team well.

branilson, 2021-09-13:
Does anyone know the procedure for porting a new board using Zephyr RTOS? I would like to do it for less expensive boards like the Black Pill V2.0 and STM Nucleo-F411RE.

pablogs9, 2021-09-13:
Have you tried using [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup) and just `ros2 run micro_ros_setup create_firmware_ws.sh zephyr [your board name in Zephyr toolchain]`?

It should try to build the library for the target platform.

branilson, 2021-09-13:
Hello, @pablogs9!

Yes, I tried. But now I see why the example apps were not working. I had to make some changes:

1. In the app folder, within the file `CMakelist.txt`, I added the board I am using. In this case, nucleo_f411re.
`set(COMPATIBLE_BOARDS disco_l475_iot1 olimex_stm32_e407 native_posix nucleo_h743zi nucleo_f746zg nucleo_f411re)`

2. In the file `firmware/zephyr_apps/microros_extensions/microros_transports.h` I change the default serial transport to the default UART "2", connected to the St-link serial in this board.
`// Set the UART used for communication. e.g. {.fd = 0} leads to UART_0 being used.
static zephyr_transport_params_t default_params = {.fd = 2};`

Another thing I noticed with this board is the necessity of resetting it manually after flashing.

I've tried successfully the ping-pong and int32_publisher apps.

Thank you!

pablogs9, 2021-09-14:
Nice to hear that @branilson, maybe you want to create a PR to contribute this couple of changes in order to provide support to this board.

branilson, 2021-09-30:
> Nice to hear that @branilson, maybe you want to create a PR to contribute this couple of changes in order to provide support to this board.

Ok, @pablogs9. I can do that. Actually, I've tested other cheap boards as well and, now I am implementing custom codes and doing other tests. Therefore, later I can contribute with suggestions of changes that result from this exploration.