micro-ROS/micro-ROS.github.io #200: Creating Firmware WS for zephyr olimex-stm32-e407 failed due to python3-pip dependency

farshbafdoustar, 2020-08-14:
Inside the [First micro-ROS Application on Zephyr](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/zephyr/) the following command for creating firmware workspace for olimex-stm32-e407 on a clean **ROS 2 Foxy FitzRoy** docker image 

`ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407`

is not successful with an error about pythoon3-pip installation

```
......
Setting up openocd (0.10.0-6build1) ...
Processing triggers for libc-bin (2.31-0ubuntu9) ...
/usr/bin/python3: No module named pip
executing command [python3 -m pip install -U west]
/usr/bin/python3: No module named pip
ERROR: the following rosdeps failed to install
  pip: command [python3 -m pip install -U west] failed
....
```

Maybe it is good to mention inside the Tutorial to install pip before creating the firmware ws.

`sudo apt-get install python3-pip`

farshbafdoustar, 2020-08-14:
That could also be added in [create_firmware_ws.sh](https://github.com/micro-ROS/micro_ros_setup/blob/8c5eb51b27dedd4797239864b4d886ae6f0b28a4/scripts/create_firmware_ws.sh#L76)

pablogs9, 2020-08-24:
Hello @farshbafdoustar, sorry for the delay.

@FranFin, did you see this problem when testing the tutorials?

FranFin, 2020-08-24:
Thanks for the tip, @farshbafdoustar. We'll update the relevant tutorials accordingly.

pablogs9, 2020-08-24:
Fixed here: https://github.com/micro-ROS/micro-ROS.github.io/pull/204

closed 2020-08-24 by pablogs9