micro-ROS/micro-ROS.github.io #203: microros foxy - Documentation

thihakyawjob, 2020-08-21:
Dear,

I have been testing microros - foxy release with stm32f4discovery board, stm32f429i-disco board. Both are failed to produce the results according to tutorial.

The issue is due to lack of documentation. I have tried to understand about MICROROS. But I still cannot make it work.

Currently OLIMEX-STM32-E407 works well with FOXY. But MICROROS team doesn't release any document about hardware implementation based on OLIMEX-STM32-E407 board. It is a biggest problem for me.

If MICROROS team releases the document about the important points to take note during microsros testing, it will be good for us.

Now stm32f4discovery board doesn't work even though STM32 chip is still the same.

I did check and test with a lot of various configuration but it still fails. It means some hidden configurations are required to make it work. But I don't know where to look for it.

I suggest MICROROS team to release an implementation procedure document for the opensource developers. Then our MIROROS developers can help you to test with various hardwares and report back the test results. Thus our community will grow faster.

Now I want to contribute a lot to Microros development but I feel that there is no complete information required by developers.

Thank you.

Regards,
Thiha Kyaw

FranFin, 2020-08-24:
Hi @thihakyawjob 

The micro-ROS project with the latest Foxy release officially supports the following boards (see [here](https://micro-ros.github.io/docs/overview/hardware/) and [here](https://github.com/micro-ROS/micro_ros_setup/pull/140)):
- Olimex LTD STM32-E407
- STM32L4 Discovery kit IoT
- Crazyflie 2.1 Drone
- ESP32

Therefore, at present, we do not support the two boards you mention above.

What tutorial are you referring to when you speak of failing results and on which RTOS?
Please notice that very complete [reference tutorials](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/) are provided on our webpage for testing a micro-ROS app on one of the three supported RTOSes (NuttX, Zephyr and FreeRTOS) on an Olimex LTD STM32-E407 evolution board.
Also notice that, in the case of Zephyr, it is relatively easy to port micro-ROS to one of the many boards supported by the Zephyr Project listed [here](https://docs.zephyrproject.org/latest/boards/index.html), among which both the  [STM32F4 Discovery](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) and the [STM32F429I Discovery](https://docs.zephyrproject.org/latest/boards/arm/stm32f429i_disc1/doc/index.html) boards can be found.

If you and your team are eager to implement the above-mentioned ports, we will be glad to be helpful at any stage of the process and discuss them with you in the relevant PRs and loops.

Best,
Francesca Finocchiaro

thihakyawjob, 2020-08-24:
Dear Francesca Finocchiaro,

Thank you for your reply. We will keep using Nuttx. Therefore please help to implement the above-mentioned ports.

We would like to support Microros. You can treat us as a part of your team. We will test it out according to your team suggestion. If your team wants to test more, we can help you also.

Personally I don't like to limit the boards for an opensource system. Therefore I want to port more boards. But we need more technical information than a working project. As you mentioned, currently MICROROS  using Nuttx is limited to above listed boards. Let open up a bit more. We will help you. We will report whatever we see during testing. It is not only for our project but also for Microros community.

I will test it using Zephyr also. Report the error as well.

Regards,
Thiha

FranFin, 2020-08-26:
Dear @thihakyawjob 

We're excited to hear that you're willing to actively participate in the micro-ROS project and give a hand! Of course, the more boards supported, the better. It would be nice to engage in direct communication in order to better align and so that we can learn more about you and your project. If interested, please contact me at francescafinocchiaro@eprosima.com.

Regards,
Francesca

closed 2020-11-11 by FranFin