micro-ROS/micro-ROS.github.io #207: Changing Webpage License from Apache 2.0 to Creative-Commons

FranFin, 2020-08-26:
Hello contributors to the [micro-ROS webpage](https://micro-ros.github.io/)!

We are migrating the license to [Creative Commons Attribution-NoDerivatives 4.0 International (CC BY-ND 4.0)](https://creativecommons.org/licenses/by-nd/4.0/), which is fitter to the content of the page than the previously employed (Apache license 2.0)[https://www.apache.org/licenses/LICENSE-2.0]. We kindly ask all external people that contributed to the webpage development to confirm that they agree with this change of license terms: @aksakalli @rpcme @coliff @rjhcnf @jayhou @andre-nguyen @alsaibie.

If not notified otherwise within a two-weeks time slot, we will proceed to the change of license.

Best regards and thanks,
Francesca Finocchiaro

andre-nguyen, 2020-08-26:
all good for me

aksakalli, 2020-08-26:
all good

rjhcnf, 2020-08-26:
Good to me.

alsaibie, 2020-09-11:
All good

ralph-lange, 2020-11-11:
@FranFin, please update the LICENSE file in the repo, too.

ralph-lange, 2020-11-11:
@FranFin, also the README.md has to be updated, where I already added a paragraph explaining that the template comes under MIT license and mentioning the commits from where on we started to add our own contents which are under the new license.

closed 2021-08-12 by ralph-lange