micro-ROS/micro-ROS.github.io #221: Agent runtime error: undefined symbol: uxr_run_session_until_data

AndyBlightLeeds, 2020-10-09:
I followed the instructions in the tutorial <https://micro-ros.github.io//docs/tutorials/core/first_application_linux/>.  Everything appeared to build correctly but, when I ran the agent, I received the error message:

```bash
$ ros2 run micro_ros_demos_rclc ping_pong
UDP mode => ip: 127.0.0.1 - port: 8888
[INFO] [1601993557.646656111] []: Created a timer with period 2000 ms.

INFO: rcl_wait timeout 1000 ms
/home/build/code/microros_ws/install/micro_ros_demos_rclc/lib/micro_ros_demos_rclc/ping_pong/ping_pong: symbol lookup error: /home/build/code/microros_ws/install/rmw_microxrcedds/lib/librmw_microxrcedds.so: undefined symbol: uxr_run_session_until_data
build@7927c5770415:~/code/microros_ws$ source /opt/ros/foxy/setup.bash
build@7927c5770415:~/code/microros_ws$ source install/local_setup.bash
build@7927c5770415:~/code/microros_ws$
build@7927c5770415:~/code/microros_ws$ # Run a micro-ROS node
build@7927c5770415:~/code/microros_ws$ ros2 run micro_ros_demos_rclc ping_pong
UDP mode => ip: 127.0.0.1 - port: 8888
[INFO] [1601994170.524389384] []: Created a timer with period 2000 ms.

INFO: rcl_wait timeout 1000 ms
/home/build/code/microros_ws/install/micro_ros_demos_rclc/lib/micro_ros_demos_rclc/ping_pong/ping_pong: symbol lookup error: /home/build/code/microros_ws/install/rmw_microxrcedds/lib/librmw_microxrcedds.so: undefined symbol: uxr_run_session_until_data
```

The error message shows that the function `uxr_run_session_until_data` is not present in the library `librmw_microxrcedds.so`.

This problem can be worked around by using two different workspaces, one for the firmware and the other for the agent as suggested by issue #141.

The commands I used to build in two workspaces were:

```bash
mkdir -p microros_firmware_ws/src
cd microros_firmware_ws/
. /opt/ros/foxy/setup.bash
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh host
ros2 run micro_ros_setup build_firmware.sh
source install/local_setup.bash
ros2 run micro_ros_demos_rclc ping_pong
```

```bash
mkdir -p microros_agent_ws/src
cd microros_agent_ws/
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
source /opt/ros/foxy/setup.bash
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
```

I have added this issue to help other people resolve the problem until a fix is implemented.

pablogs9, 2020-10-09:
Hello @AndyBlightLeeds, thank you very much for this comment. We are aware of this error because the agent is downloading its own version of the Client library for P2P issues and the versions differs: [check here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/a34ad20a6dcc457cf34a3df97cacb87b00a2ff9d/CMakeLists.txt#L77).

We are working on a solution to align situations when two clients libraries are installed in the same workspace, we will alert you here when this is solved. 

Thank you very much again for the tip.

@jamoralp

pablogs9, 2020-10-13:
This should be solved with https://github.com/micro-ROS/micro_ros_setup/pull/211.

Please reopen if you have any problem.

AndyBlightLeeds, 2020-10-13:
@pablogs9  I have built the Linux host tutorial in a clean docker and workspace using the latest version and confirm that the problem has been fixed.  Thanks for taking the time to fix this problem.

pablogs9, 2020-10-13:
Thank you for alerting us!