micro-ROS/micro-ROS.github.io #224: Cannot trace pingpong on OlimexE407-_FreeRTOS

Aerotic, 2020-10-21:
I'm so sorry. I don't know a proper repo where to submit this issue.

HOST OS: Ubuntu 20.04.1 LTS

used command for tracing: 
```shell
ros2 trace
```
error log:
```shell
aero@aero:~/ros/freertos$ ros2 trace
UST tracing enabled (16 events)
kernel tracing enabled (4 events)
context (3 names)
writing tracing session to: /home/aero/.ros/tracing/session-20201021211759
press enter to start...
channel enabling failed: Tracing the kernel requires a root lttng-sessiond daemon, as well as "tracing" group membership or root user ID for the lttng client.
```
 I'v done everything mentioned in the tutorial https://micro-ros.github.io/docs/tutorials/advanced/tracing
but i'm stuck at this step `ros2 launch tracetools_launch example.launch.py`, which report similar error about  `Tracing the kernel requires a root lttng-sessiond daemon, as well as "tracing" group membership or root user ID for the lttng client`

christophebedard, 2020-10-21:
> Yet, one detail is not mentioned in the tutorial: You have to add your user to the group _tracing_ (by `sudo usermod -aG tracing $USER`). Please note that you have to log off and log in again to update the group memberships

This is mentioned in the "setup" section of the tutorial.

christophebedard, 2020-10-21:
Wait, Ralph just added it, nevermind :sweat_smile:

However, this is something that is part of LTTng's installation steps, which the tutorial links to (https://lttng.org/docs/#doc-ubuntu-ppa). We can't really just copy paste all the steps from LTTng's documentation.

Aerotic, 2020-10-22:
> Hi @Aerotic, thank you for reaching out. I just tried the tutorial on a fresh Ubuntu 20.04 installation with ROS 2 Foxy and it worked for me immediately. Yet, one detail is not mentioned in the tutorial: You have to add your user to the group _tracing_ (by `sudo usermod -aG tracing $USER`). Please note that you have to log off and log in again to update the group memberships before calling `ros2 launch tracetools_launch example.launch.py` again.
> Does this solve the issue?

Your solution half-solves this issue “:’-D”. It works that I run the cmd `ros2 launch tracetools_launch example.launch.py` as root. But there is the same error when I run the cmd as aero (my own user). Both root and $USER are added to the `tracing` group. Here is the log of `cat /etc/group`:
```shell
aero@aero:~$ cat /etc/group
root:x:0:
daemon:x:1:
bin:x:2:
sys:x:3:
adm:x:4:syslog,aero
tty:x:5:syslog
disk:x:6:
lp:x:7:
mail:x:8:
news:x:9:
uucp:x:10:
man:x:12:
proxy:x:13:
kmem:x:15:
dialout:x:20:
fax:x:21:
voice:x:22:
cdrom:x:24:aero
floppy:x:25:
tape:x:26:
sudo:x:27:aero
audio:x:29:pulse
dip:x:30:aero
www-data:x:33:
backup:x:34:
operator:x:37:
list:x:38:
irc:x:39:
src:x:40:
gnats:x:41:
shadow:x:42:
utmp:x:43:
video:x:44:
sasl:x:45:
plugdev:x:46:aero
staff:x:50:
games:x:60:
users:x:100:
nogroup:x:65534:
systemd-journal:x:101:
systemd-network:x:102:
systemd-resolve:x:103:
systemd-timesync:x:104:
crontab:x:105:
messagebus:x:106:
input:x:107:
kvm:x:108:
render:x:109:
syslog:x:110:
tss:x:111:
bluetooth:x:112:
ssl-cert:x:113:
uuidd:x:114:
tcpdump:x:115:
avahi-autoipd:x:116:
rtkit:x:117:
ssh:x:118:
netdev:x:119:
lpadmin:x:120:aero
avahi:x:121:
scanner:x:122:saned
saned:x:123:
nm-openvpn:x:124:
whoopsie:x:125:
colord:x:126:
geoclue:x:127:
pulse:x:128:
pulse-access:x:129:
gdm:x:130:
lxd:x:131:aero
aero:x:1000:
sambashare:x:132:aero
systemd-coredump:x:999:
rdma:x:133:
tracing:x:134:aero,root

```

iluetkeb, 2020-10-22:
> But there is the same error when I run the cmd as aero (my own user).

Is it exactly the same error? There are also other permissions issues that may crop up, for example, you may need to change some options in the kernel through sysctl.

A good way to debug this is to use to `lttng create && lttng enable-event -k -a`. If that works, you're good to go. If it doesn't, it will print helpful information on which sysctl setting you have to change (at least in the most recent 2.12 version of LTTng).

> Is it the right way that I must run as root?

I would recommend against it. It's harder to integrate with ros that way (you definitely should not run your ROS system as root) and generally, avoid root as much as possible.

Aerotic, 2020-10-22:
> Is it exactly the same error? There are also other permissions issues that may crop up, for example, you may need to change some options in the kernel through sysctl.
>
> A good way to debug this is to [...]

Thanks! It works well!
Also, your solution solves part of the issue #226 
When I run as root, it werid that the trace works but cannot identify which is the used `example.launch.py `

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9