micro-ROS/micro-ROS.github.io #226: Cannot find the proper launch file

Aerotic, 2020-10-22:
according to  `Callback duration analysis` (Callback duration analysis) I should edit launch file in the path `ros2_ws/src/ros2/tracing/tracetools_launch/launch/`
I've finished previous steps, but I cannot find the directory `ros2_ws/src/ros2`. There are only two directories, `ros-tracing` and `tracetools_analysis` , in the path `ros2_ws/src/`
By the way, does the launch file refer to `example.launch.py` ?  I've found multiple `example.launch.py` :
```shell
./install/tracetools_launch/share/tracetools_launch/launch/example.launch.py
./build/tracetools_launch/launch/example.launch.py
./src/ros-tracing/ros2_tracing/tracetools_launch/launch/example.launch.py
```
which one is used for the cmd `ros2 launch tracetools_launch example.launch.py`

ralph-lange, 2020-10-22:
I just fixed the path in the tutorial. Please edit the file in the src folder and then call `colcon build` and `ros2 launch tracetools_launch example.launch.py` again.

christophebedard, 2020-10-22:
> ./install/tracetools_launch/share/tracetools_launch/launch/example.launch.py
./build/tracetools_launch/launch/example.launch.py
./src/ros-tracing/ros2_tracing/tracetools_launch/launch/example.launch.py
which one is used for the cmd ros2 launch tracetools_launch example.launch.py

It's all the same file. You should only modify the one in your `src/` directory. The other ones are created/copied when you build your workspace.

Aerotic, 2020-10-23:
> It's all the same file. You should only modify the one in your `src/` directory. The other ones are created/copied when you build your workspace.

got it. thanks! :+1:

christophebedard, 2020-11-06:
Looks like this can be closed.

closed 2020-11-06 by christophebedard