micro-ROS/micro-ROS.github.io #228: how to trace micro-ROS

Aerotic, 2020-10-23:
I've finished all steps mentioned in https://micro-ros.github.io/docs/tutorials/advanced/tracing/
But there exists a question how to trace `pingpong` demo over FreeRTOS using OlimexE407 .
I've read docs in https://gitlab.com/ros-tracing/ros2_tracing ,
but honestly, the docs talk more about the design rather than the manual that teaches developers  how to trace.

christophebedard, 2020-10-24:
I personally have never tried tracing on any of the hardware that Micro-ROS supports.

Can you be a bit more specific? The tracing tutorial shows an example. What else are you looking for, and what have you tried?

Aerotic, 2020-10-25:
> I personally have never tried tracing on any of the hardware that micro-ROS supports.
> 
> Can you be a bit more specific? The tracing tutorial shows an example. What else are you looking for, and what have you tried?

Specifically, I'd like to trace how microROS works in `pingpong`  demo on OlimexE407. I can use debugger and openocd to trace the code step by step , but something may not be so convenient to trace using that way, e.g. callback duration.
For now, I've tried all steps mentioned in https://micro-ros.github.io/docs/tutorials/advanced/tracing/.  And I've modified ` example.launch.py` like this:
```python
Node(
            package='tracetools_test',
            executable='microROS/pong',
            arguments=['do_more'],
            output='screen',
        ),
```
But it reports errors:
```shell
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=SubstitutionFailure("executable 'microROS/ping' not found on the libexec directory '/home/aero/ros/freertos/install/tracetools_test/lib/tracetools_test' ")
```
What can I trace on the OlimexE407 and how should I do that?

christophebedard, 2020-10-25:
> `executable='microROS/pong',`

Unfortunately that's not really how it works. This is trying to run an executable called `microROS/pong` in the `tracetools_test` package. Is the `pingpong` demo a node? Is it a combination of nodes?

> but honestly, the docs talk more about the design rather than the manual that teaches developers how to trace.

Aerotic, 2020-10-25:
> > `executable='microROS/pong',`
> 
> Unfortunately that's not really how it works. This is trying to run an executable called `microROS/pong` in the `tracetools_test` package. That doesn't exist.
> 
> Is the `pingpong` demo a node? Is it a combination of nodes?
> 
> > but honestly, the docs talk more about the design rather than the manual that teaches developers how to trace.
> 
> The two ways that you can use to trace are introduced here: https://gitlab.com/ros-tracing/ros2_tracing#tracing

The `pingpong` demo is introduced in [First micro-ROS Application on FreeRTOS](https://micro-ros.github.io//docs/tutorials/core/first_application_rtos/freertos/)

Aerotic, 2020-10-25:
> Unfortunately that's not really how it works. This is trying to run an executable called `microROS/pong` in the `tracetools_test` package. That doesn't exist.
>
> Is the `pingpong` demo a node? Is it a [...]

I'd know that I need to specify `package` and `executable`. But how to do that is not mentioned in the doc about trace you recommended:blush:​

Aerotic, 2020-10-25:
> Unfortunately that's not really how it works. This is trying to run an executable called `microROS/pong` in the `tracetools_test` package. That doesn't exist.
>
> Is the `pingpong` demo a node? Is it a [...]

You said that
>  Is the pingpong demo a node? Is it a combination of nodes?
I used to regard the program running on OlimexE407 as a microROS node, isn't it?
If it's a ROS node, how can I trace it? 
If not, is that mean the program is more likely a MicroXCREDDS-client rather than a ROS node?

christophebedard, 2020-10-26:
Just to make it clear in case it wasn't already, the tracing stuff, as it is right now, is not really meant to work on the hardware that micro-ROS supports. While you may be able to get it working using other methods if you try, the workflow (for tracing/building/executing) is very different compared to an amd64 machine (which is what the tracing tutorial was aimed at). So the tools for tracing (`ros2 trace` and the `Trace` action for launch files) most definitely won't work.

Sorry if this wasn't clear.

Aerotic, 2020-10-27:
> Just to make it clear in case it wasn't already, the tracing tools, in their current state, are not really meant to work on the hardware that micro-ROS supports. While you may be able to get it working using other methods if you try, the workflow (for tracing/building/executing) is very different compared to an amd64 machine (which is what the tracing tutorial was aimed at). So the tools for tracing (`ros2 trace` and the `Trace` action for launch files) most probably won't work.
> 
> If you can set up LTTng on one of the supported platforms (not sure if they support it), then you can try enabling tracing with LTTng directly instead of using the tracing tools.
> 
> Sorry if this wasn't clear.

I got it, thanks for reply :blush:

closed 2022-01-28 by pablogs9