micro-ROS/micro-ROS.github.io #234: Rework webpage

FranFin, 2020-11-11:
Tasks regarding the webpage restructuring as discussed during the OFERA f2f meeting of 14.10.2020

- [ ] Landing Page
    - [ ] Add a list of main features @ralph-lange 

- [ ] Overview section

    - [ ] Architecture
        - [ ] Update/rework Architecture diagram @ralph-lange 
        - [ ] Link each layer to associated features, according to this [document](https://docs.google.com/document/d/1I-bDoI5YEKSdL-dbM8IDUghDALIzEOHCppiAPUIvC-A/edit) @ralph-lange 
        - [ ] Remove ‘Source Code Repositories’ part, and move it to the ‘Docker and CI status’ @FranFin 

    - [ ] Feature Status
        - [ ] Now meant to represent comparison with features of ROS 2: change name accordingly
        - [ ] Add/delete features. Missing:
            - Actions
            - Graphs
            - Parameters
            - Discovery
            - QoS?
            - p2p (or explain difference between new p2p and CED middleware)
            - Operations within a ROS 2 node
            - More: Support for RT code - add sentence regarding rclpy and rclcpp
        - [ ] Add third column for symbols that encode features status [Done; Done partially; WIP; Not done but envisioned in the very near future; Not done and not planned] + Legend
        - [ ] Split table into sub-tables: group features by category that +- represent the features list presented in 'Landing page' and ‘Architecture’. Small headings in bold between categories.
        - [ ] Group features that are not and won’t be implemented at the end of the table

    - [ ] Supported Hardware
        - [ ] Change appearance: only logos, and then you get the info when clicking on the logo
        - [ ] Add sentence explaining that more boards have been ported by users and link to table in micro_ros_setup README

    - [ ] Supported RTOSes [page to be created]: Add RTOSes in the same form as the hardware, with clickable logos with short description (while removing them from ‘Concepts’ page)

    - [ ] Transport and Data Links: Change the table to be more similar (or even equal) to the one that is found in the micro_ros_setup README (specifying which transports are allowed for each available combination of HW and RTOS)

    - [ ] Docker and CI status
        - [ ] Rename as ‘Repositories, Dockers and CI status’
        - [ ] Add ‘Source Code Repositories’ part from ‘Architecture’, and update it
        - [ ] Update Docker and CI status part

    - [ ] Members, partners and users
        - [ ] Add new partners and MW users
        - [ ] Add 'Collaborators' section

    - [ ] Licensing [page to be created]: Add category explaining micro-ROS' licenses

- [ ] Concepts section
    - [ ] 

Other tasks:
- [ ] 
- [ ]

ralph-lange, 2020-11-11:
The new feature list is now online at https://micro-ros.github.io//docs/overview/features/ with a collapsible layout. However, without any graphics the page is a bit boring. Two options:
1. Just add some graphics of an MCU to brighten up the page and create a separate page for the architecture overview.
2. Include the architecture diagram, but _below_ the list of features. I may squeeze the list of features and even make it a [accordion](https://jqueryui.com/accordion/). Of course, the page should be named "Features and Architecture" then.

What do you think, @FranFin?

FranFin, 2020-11-12:
@ralph-lange I like it a lot! Maybe the formatting is too tight. Would it be possible to make it a little bit bigger?

ralph-lange, 2020-11-12:
> 
> 
> @ralph-lange I like it a lot! Maybe the formatting is too tight. Would it be possible to make it a little bit bigger?

Done in https://github.com/micro-ROS/micro-ROS.github.io/commit/25c0051e37c8e91e1bb6df3925184f9ff8229b0b.

ralph-lange, 2020-11-12:
@FranFin, regarding "Add third column for symbols that encode features status": I already added this column in https://github.com/micro-ROS/micro-ROS.github.io/pull/232/commits/aed04becee685527c64051934143e465b2ff2d10.

closed 2021-08-12 by ralph-lange