micro-ROS/micro-ROS.github.io #240: How to use ROS parameters with microROS ?

cocodmdr, 2020-11-23:
Hello,

I have read the tutorial **Programming with rcl and rclc**, but it does not mention parameters.
I would like to use ROS parameter in order to pass them to the microROS application. Is there a way to do that ?

Here are the parameters tutorial in ROS2 that I would like to use with my microROS application.
[https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/](url)
[https://index.ros.org/doc/ros2/Tutorials/Using-Parameters-In-A-Class-CPP/](url)

Regards,
Cocodmdr

ralph-lange, 2020-11-23:
Hello @cocodmdr, the parameters API for rclc is still under development. You may try out a first version at https://github.com/micro-ROS/rclc/tree/feature/add_parameters. Once the corresponding PR https://github.com/micro-ROS/rclc/pull/39/ is merged, we will extend the mentioned tutorial.

Feedback to the parameters API is highly appreciated! Just comment on https://github.com/micro-ROS/rclc/pull/39/, please.

pablogs9, 2020-11-24:
Closing since solved

closed 2020-11-24 by pablogs9