micro-ROS/micro-ROS.github.io #246: A bug may caused by the gcc version?

zijian1998, 2020-11-25:
about tutorials  .
The OS is Ubuntu 20.04.1
The gcc version is 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) 

When I followed the tutorials 'First micro-ROS Application on Linux', and tried to create the micro-ROS agent , an error happened.
```
.../microros_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/utils/ArgumentParser.cpp:24:14: error: ‘bool eprosima::uxr::agent::parser::utils::usage(const char*, bool)’ should have been declared inside ‘eprosima::uxr::agent::parser::utils’
   24 |  bool no_help)
      |              ^
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:310: CMakeFiles/microxrcedds_agent.dir/src/cpp/utils/ArgumentParser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_agent.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
So I checked the cpp and the hpp,and I'm sure that the function named usage have been declared inside the namespace.
```
bool eprosima::uxr::agent::parser::utils::usage(
	const char* executable_name,
	bool no_help)
{
    std::stringstream ss;
    ss << "Usage: '" << executable_name << " <udp4|udp6|tcp4|tpc6";
#ifndef _WIN32
    ss << "|serial|pseudoterminal";
#endif // _WIN32
    ss << "> <<args>>'" << std::endl;
    if (no_help)
    {
        ss << "For a more detailed description about all the available arguments, ";
        ss << "please execute the agent with '-h/--help' option." << std::endl;
    }
    std::cout << ss.str();
    return false;
}
```
```
bool usage(
        char* executable_name,
        bool no_help = true);

TransportKind check_transport(
        const char* transport);

Middleware::Kind get_mw_kind(
        const std::string& kind);
} // namespace utils
```
However the first param of the function declared in the namespace is "char*" not the "const char*".So I solve it by add the 'const' before the 'char* executable_name' in the namespace.
So is it a bug caused by the gcc version?   
This is my first time to have an issue.I hope I didn't bother you.
Thank you

jamoralp, 2020-11-25:
Good morning,
This is related with https://github.com/micro-ROS/micro_ros_setup/issues/237. It has been fixed already. Sorry for the inconvenience. Indeed, the signature between the function's declaration and implementation did not match, so that was causing the error.

jamoralp, 2020-11-25:
please update your Micro-XRCE-DDS-Agent repository, under the foxy branch, or delete and re-create the agent workspace. Unless you have any other question, I'm closing this issue, since solved.

closed 2020-11-25 by jamoralp