micro-ROS/micro-ROS.github.io #250: Update CI-build status ros2/rclc

JanStaschulat, 2020-12-02:
@ralph-lange @pablogs9
I would like to have a status page for the releases of the packages in ros2/rclc:
The overview on https://micro-ros.github.io/docs/overview/docker_ci_status/ is only for the complete repository and only for arm64 architecture:

packages:

- rclc

- rclc_lifecycle

- rclc_examples

architectures:

- amd64

- armhf

- arm64

releases:

- Dashing

- Eloquent

- Foxy

- Rolling

Where could I create such a page?

- at micro-ros.github.org

- at ros2/rclc

What do you think?

ralph-lange, 2020-12-02:
I propose to add it at both places: in the README.md in ros2/rclc as well as on the existing page on micro-ros.github.org. For the latter, a new section should be added given the different architectures, etc.

jamoralp, 2020-12-02:
Hello @JanStaschulat, I recently opened a PR for reworking the CI status section. Maybe you could address these changes there? #248

FranFin, 2020-12-02:
I back up the idea of adding the statuses for the releases of the `rclc` packages in the PR that @jamoralp is mentioning as well as in the README of the ros2/rclc repo.

JanStaschulat, 2020-12-03:
Okay, I will add the table to  PR #248

JanStaschulat, 2020-12-03:
created https://github.com/micro-ROS/micro-ROS.github.io/pull/253

pablogs9, 2022-01-28:
All merged, I'm closing, reopen if needed

closed 2022-01-28 by pablogs9