micro-ROS/micro-ROS.github.io #254: Releases-status of micro-ros/rclc

JanStaschulat, 2020-12-03:
@ralph-lange @jamoralp 

In http://localhost:4000/docs/overview/docker_ci_status/
the rclc repository is listed only with Dashing release. 

The master branch or rclc is compliant to Eloquent and Foxy as well.
What do you suggest? 

we wanted to avoid multiple branches in rclc repository so far. So I propose to add 
- line for Eloqent release links to rclc:master
- line for Foxy release links to rclc:master

I would like to avoid, that people think, that rclc only works for ROS 2 Dashing and does not support Eloquent/Foxy because it is not listed in this table.

ralph-lange, 2020-12-03:
Hi @JanStaschulat, with "line for Eloquent" and "line for Foxy" you mean a second and third badge labelled "ROS: Eloquent" and "ROS: Foxy" referring to the same job for the master branch. Yes, that's a good idea!

JanStaschulat, 2020-12-04:
@ralph-lange  Yes, that was my idea.

jamoralp, 2020-12-04:
Ok, sounds reasonable to me. I'm doing this in a PR we opened to fix several issues with the reworked CI page #255

pablogs9, 2022-01-28:
Can we close?

closed 2022-01-28 by JanStaschulat