micro-ROS/micro-ROS.github.io #303: micro_ros_agent package failed

shic15, 2021-04-03:
- Installation type:  micro_ros_setup
- Version: foxy in Ubuntu 20.04.2
While following the basic tutorials from [https://micro.ros.org//docs/tutorials/core/first_application_linux/] all the steps executed fine until I reached the command: `ros2 run micro_ros_setup build_agent.sh`

#### Steps to reproduce the issue
After installing ROS2 I followed the "First application Linux" tutorials as it is and the package failure occurred while building micro ros agent.

#### Expected behavior
The agent should build successfully

#### Actual behavior
`Building micro-ROS Agent
Starting >>> microcdr
Starting >>> builtin_interfaces
Starting >>> microxrcedds_agent
Finished <<< microcdr [0.78s]                                                                                                     
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [2.15s]                                                                                          
Starting >>> micro_ros_msgs
Finished <<< builtin_interfaces [10.0s]                                                                                                        
Finished <<< micro_ros_msgs [9.05s]                                                                         
[Processing: microxrcedds_agent]                             
[Processing: microxrcedds_agent]                                      
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
--- stderr: microxrcedds_agent                                         
Cloning into 'spdlog'...
Cloning into 'sanitizers'...
Already on 'master'
Note: switching to 'v1.4.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedhead to false

HEAD is now at 1549ff12 Replace STRING(PREPEND ..) in CMakeLists.txt to support older CMake versions
---
Finished <<< microxrcedds_agent [5min 3s]
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                     
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `spdlog::logger::log_it_(spdlog::details::log_msg const&, bool, bool)'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/shic/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:173: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [24.5s, exited with code 2]

Summary: 5 packages finished [5min 28s]
  1 package failed: micro_ros_agent
  2 packages had stderr output: micro_ros_agent microxrcedds_agent
`
#### Additional information
I could not find and similar issue, although I tried to pull the eProsima/Micro-XRCE-DDS-Agent repo and tried building again using `colcon build` still the same issue.

I'm relatively new to ROS and do not have much knowledge regarding it, pardon me if I had missed something,
Thanks in advance.

pablogs9, 2021-04-05:
Related: https://github.com/micro-ROS/micro-ROS-demos/issues/43

pablogs9, 2021-04-05:
Which compiler are you using `gcc` or `clang`?

shic15, 2021-04-05:
Default ROS2 compiler, gcc.

pablogs9, 2021-04-05:
I have followed the tutorial instructions and it builds seamlessly:

https://asciinema.org/a/cHjRD383IpX5EJxx1H8rsJmFx

shic15, 2021-04-05:
That's is weird, even I tried installing micro ros thrice after completely deleting the workspace.

Could the results be different cause you're using docker? or is the problem somehow related with ROS2 installation?

pablogs9, 2021-04-05:
Could you share the files inside `log` folder for micro xrce dds agent and micro ros agent?

shic15, 2021-04-05:
Sure, I've attached the zip file of the latest build in the log folder.

[uros_log.zip](https://github.com/micro-ROS/micro-ROS.github.io/files/6259186/uros_log.zip)

pablogs9, 2021-04-06:
Please, delete the folders `build` `install` and `log` and rebuild the micro-ROS agent and then attach here the log folder.

shic15, 2021-04-06:
I rebuild everything from scratch as I was getting some errors. This is a fresh install and build.
[log.zip](https://github.com/micro-ROS/micro-ROS.github.io/files/6262747/log.zip)

pablogs9, 2021-04-06:
I have seen that you have a custom installation of ROS 2 Foxy in `/home/shic/ros2_foxy/ros2-linux`.

Does this problem happen with a ROS 2 installation from binaries with `apt`?

Do you have installed `sudo apt-get install -y libfmt-dev`?

shic15, 2021-04-06:
I didn't make any customizations while installing, directly installed ROS2 from [here](https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html), and don't think I got a similar error there.
Are the instructions appropriate for learning purposes? There is another page for installation [here](https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Binary.html), should I uninstall and proceed with that?

The fmt library is already installed.

shic15, 2021-04-07:
I did try reinstalling ros2 but still no help. Same error occours.

pablogs9, 2021-04-08:
Well, since this is not a reproducible issue, do you have any inconvenience in using any of the other options:
- the dockerized version of the agent: `docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6`
- the snap pacakge: https://snapcraft.io/micro-ros-agent

shic15, 2021-04-08:
Not really, I'm ready to use whatever works. Just that I'm a newbie to this so approached the solution which had simple documentation.

To use the snap package I need to follow the [tutorials](https://micro.ros.org//docs/tutorials/core/first_application_linux/) till building the firmware
 ```
ros2 run micro_ros_setup build_firmware.sh
source install/local_setup.bash
```
, then use [snap package](https://snapcraft.io/micro-ros-agent) commands for creating and building the agent, right?

pablogs9, 2021-04-08:
I prefer the docker version, but the snap package should work, just follow the instructions in the snap webpage

shic15, 2021-04-08:
Finally got lucky with the dockerized micro_ros_agent. Also TIL about and it is pretty crazy what you can do with it, zero environment setup efforts full plug and play. Thanks for the help.

closed 2021-04-09 by pablogs9