micro-ROS/micro-ROS.github.io #308: microros service bridge to ROS1

CyrilNeoFarm, 2021-05-04:
- Hardware description: ESP32 DevKitC
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version: Foxy for ROS2 and Noetic for ROS1

The ros1_bridge doesn't bridge services coming from micro-ros.

#### Steps to reproduce the issue

-build the bridge following these instructions: https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst
-compile/build/flash the server node
-run the agent
-run the bridge: `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`

#### Expected behavior

The bridge should detect the micro-ros service, and I should be able to call it from ROS1.

#### Actual behavior

The bridge doesn't seem to detect services coming from micro-ros.

#### Additional information

The bridge works when I'm running a server node on ros2.
The bridge works when I'm running publishers on micro-ros.
The bridge doesn't work when the server node is on micro-ros event though I can access the service from ROS2 without issue.

pablogs9, 2021-05-04:
Could you specify the version of the micro-ROS agent that you are using?

As far as I have understood you are trying to use the following setup:

```
micro-ROS Service Server <----> ROS1 Bridge <----> ROS 1 Service Client
```

Is that correct? Does the opposite works?

```
micro-ROS Service Client <----> ROS1 Bridge <----> ROS 1 Service Server
```

CyrilNeoFarm, 2021-05-04:
Micro-ROS agent version : 0.0.1
Yes that is exactly what I am trying to do. I have not yet tried to do the opposite though.

pablogs9, 2021-05-04:
Could you test your scenario with our latest micro-ROS dockerized version:

```
docker run -it --rm --net=host -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

If this agent version has the same behavior, we will try to check what is happening in this scenario.

closed 2021-05-10 by CyrilNeoFarm