micro-ROS/micro-ROS.github.io #412: ISOLATION from different systems

Muthukumar4796, 2023-12-08:
HI, i am working with ros2 and micro-ros but I don't want to share my topics within the same network so I kept ROS_DOMAIN_ID=10 in my bashrc but while running micro ros agent I am not getting my topics. its because of ros_domain_id is not allowing to communicate between micro-ros and ros2 .

so i want a solution to communicate between ros2 and micro-ros and also i don't want to share my instances within the same network.

can anyone help me in this?
Thankyou.

pablogs9, 2023-12-11:
Are you setting the Domain ID in the micro-ROS client side?

Muthukumar4796, 2023-12-11:
I'm not entirely sure what you mean by client-side. Are you referring to the micro ROS workspace?

pablogs9, 2023-12-11:
In your micro-ROS application, are you setting the domain id to 10 as you can see here:  https://github.com/micro-ROS/micro-ROS-demos/blob/70f3cbf27ce12dc9e57e5095b4233553e207d866/rclc/configuration_example/configured_publisher/main.c#L49

closed 2025-09-16 by 4ntn